DK9BT/esp32-e22-lora-board #5: SCH V 2.0.6 & PCD V 2.0.1

karamo, 2024-10-04:
[esp32-e22-lora-board_2.0.6.zip](https://github.com/user-attachments/files/17263653/esp32-e22-lora-board_2.0.6.zip)

**SCH V 2.0.6** ist so weit mal fertig  
[esp32-e22-lora-board_2.0.6.pdf](https://github.com/user-attachments/files/17263654/esp32-e22-lora-board_2.0.6.pdf)

**PCB V 2.0.1** steht zur Diskussion  
[image]

3D-Modell v. E22  
[E22-400M30S-3D.zip](https://github.com/user-attachments/files/17263671/E22-400M30S-3D.zip)

karamo, 2024-10-04:
von Ben:
Ich frag mich ob wir noch Richtung Gehäuse mal schauen sollten? Irgendwas günstiges schönes...wenn die Bohrlöcher für Gehäuse XY schon passen dann hilft das dem einen oder anderem OM sicher....dann muss die Platine auch net im Müllbeutel oder Brotdose außen am Fenster hängen 😂

karamo, 2024-10-04:
Ja, sucht was aus. PCB kann ich ja dementsprechend anpsssen.

karamo, 2024-10-04:
für den SMA-Buchsen Anschluss siehe hier:

https://www.mikrocontroller.net/topic/218948

https://www.mantaro.com/index.php/resources/impedance-calculator.htm

karamo, 2024-10-05:
offene Punkte:
• 3D Modelle von den Modulen u.a. Bauteilen.
• und mit den Leiterbahnbreiten bin ich auch noch nicht überall zufrieden.
• spezielle Platzierung der I²C (2 rechts hinten) ist auch nur mal so.
• Befestigungsbohrungen fehlen natürlich auch noch.
• vorverbundene Jumper für I²C Display in default pinning

karamo, 2024-10-05:
OLED
https://github.com/pforrmi/KiCad-SSD1306-128x64
https://www.snapeda.com/parts/DM-OLED096-636/Display%20Module/view-part/
https://forum.kicad.info/t/oled-display-1x4-header-and-four-soldered-threaded-standoff-footprint-versus-bom/46372
https://github.com/TG9541/kicad-footprints
https://www.youtube.com/watch?v=iLqMU6PnAjU

karamo, 2024-10-05:
SMA
https://kicad.github.io/footprints/Connector_Coaxial
https://www.snapeda.com/parts/221789-1/TE%20Connectivity/view-part/
https://www.snapeda.com/parts/CON-SMA-EDGE-S/RF%20Solutions/view-part/

karamo, 2024-10-05:
ESP32 DevKit V4
https://www.snapeda.com/parts/ESP32-DEVKITC-32D/Espressif%20Systems/view-part/
https://grabcad.com/library/esp32-devkitc-v4-1

karamo, 2024-10-05:
OneWire benötigt einen 4k7 Ω Widerstand
[image]

karamo, 2024-10-05:
Schottky-Dioden & Schutzdioden am Versorgungsspannungseingang USB-C & Schraubklemme vorsehen.
[image]

DK9BT, 2024-10-06:
v2.0.6 commited to git

closed 2024-10-06 by karamo