DK9BT/esp32-e22-lora-board #7: SCH V 2.0.7 & PCB V 2.0.2

karamo, 2024-10-06:
[esp32-e22_SCH-2.0.7.pdf](https://github.com/user-attachments/files/17269866/esp32-e22_SCH-2.0.7.pdf)
[esp32-e22_PCB-2.0.2_3D.pdf](https://github.com/user-attachments/files/17269867/esp32-e22_PCB-2.0.2_3D.pdf)
[esp32-e22_SCH-2.0.7_PCB-2.0.2.zip](https://github.com/user-attachments/files/17269871/esp32-e22_SCH-2.0.7_PCB-2.0.2.zip)

karamo, 2024-10-06:
**offene Punkte:**
• Leiterbahnführung & Leiterbahnbreiten optimieren.
• 50Ω Stripline für SMA-Anschluss
• spezielle Platzierung der I²C-Verbinder für BME & Reserve.
• Versorgungen mit Schutzdioden siehe Kommentar.
• Befestigungsbohrungen für Gehäuse & OLED

**von Ben:**
Ich frag mich ob wir noch Richtung Gehäuse mal schauen sollten? Irgendwas günstiges schönes...wenn die Bohrlöcher für Gehäuse XY schon passen dann hilft das dem einen oder anderem OM sicher....dann muss die Platine auch net im Müllbeutel oder Brotdose außen am Fenster hängen 😂

• Wasserdichte Anschlussdose, IP65 Elektrische Projektbox Verteilerdose Staubdicht Wetterfest, ABS Kabelgehäuse Verteilerkasten für Indoor Outdoor Elektronikkabel (Transparent 15,8 x 9 x 6,4 cm)  
[https://amzn.eu/d/4mfVIbA](https://amzn.eu/d/4mfVIbA)

• andere Gehäuse
Aussengehäuse http://www.schubert-gehaeuse.de/aussengehaeuse.html

Weissblechgehäuse http://www.schubert-gehaeuse.de/weissblechgehaeuse.html

karamo, 2024-10-06:
Schottky-Dioden & Schutzdioden am Versorgungsspannungseingang USB-C & Schraubklemme vorsehen.
Als Vorlage dient der Versorgungsteil am ESP32-Board.
[image]

https://www.snapeda.com/parts/LESD5D5.0CT1G/Leshan%20Radio%20Co./view-part/
https://de.aliexpress.com/i/1005005640352091.html

https://www.snapeda.com/parts/1N5817/Taiwan%20Semiconductor/view-part/

https://www.snapeda.com/parts/BAT760-7/Diodes%20Inc./view-part/

DK9BT, 2024-10-07:
commited to git

closed 2024-10-08 by karamo