DK9BT/esp32-e22-lora-board #10: SCH V 2.0.9 & PCB V 2.0.4

karamo, 2024-10-07:
SCH als PDF:
[esp32-e22_SCH-2.0.9.pdf](https://github.com/user-attachments/files/17284116/esp32-e22_SCH-2.0.9.pdf)
KiCad Projekt:
[esp32-e22_SCH-2.0.9_PCB-2.0.4.zip](https://github.com/user-attachments/files/17284120/esp32-e22_SCH-2.0.9_PCB-2.0.4.zip)
PCB als 3D-PDF:

karamo, 2024-10-07:
**Offene Fragen / Probleme, die in der nächsten Version behoben werden:**

• das 3D-Modell der SMA-Bu ist bei der Modifikation verschwunden.
• In der SCH fehlen noch die Typen der Schutzbeschaltung.
• Die Frage ist auch, ob der USB-C Stecker wegen der größeren Ausnehmung rechts vorne vom Gehäuse überhaupt einsteckbar ist.

• Gehäusedaten in Doku

karamo, 2024-10-08:
**Erläuterungen zum Gehäuseeinbau:**

Das Gehäuse verfügt über Ø3,2x4 Sacklöcher. Darin können M2x4x3,5 Einpressmutter eingeschmolzen werden.
Bei anderen Bedarf können die Befestigungsbohrungen auf der PCB aufgebohrt werden.
[image]

Durch die stehende SMA-Buchse ist eine mechanische Entkopplung von Gehäuse-PCB-Montage zu einer ev. am Gehäuse montierter Antennen-Buchse gegeben.

Das Breakout Board mit der stehenden USB-C Buchse ermöglicht ein leichtes Anstecken der externen Versorgung.
https://de.aliexpress.com/item/1005007821332638.html

closed 2024-10-09 by karamo

HelmiBeh, 2024-10-09:
Ich würde in die Leitung vom Pin21 vom E22 zur Antennenbuchse noch einen 0-Ohm-SMD-Widerstand (0805) einbauen. Dann kann man Leitung bei Verwendung der u-Fl-Buchse abtrennen. Bei Verwendung der SMA reicht ein kleiner Lötklecks.

karamo, 2024-10-10:
> Wenn man die Bauteile in SMD machen würde, könnte man das Kleinvieh gleich beim Leiterplattenhersteller bestücken lassen. Und 1206 oder 0805 kann man auch im Alter noch gut per Hand löten.

wäre zwar zu überlegen, aber die PCB schaut auch mit den bedrahteten Bauteilen jetzt sehr aufgeräumt aus und es ist (noch) genügend Platz dafür.  
die USB-C als SMD ist auch bereits verschwunden und sind nur 4 Pins, an die man wahlweise verschiedene Teile anschließen kann. Infos hierzu folgen in den nächsten Tagen.  
Somit verbleibt nur optional die eine Transient Supressor Diode als SMD.