DK9BT/esp32-e22-lora-board #11: Use of this design with rnode and Meshchat

pa2eon, 2025-02-02:
Dears,

Found your very nice hardware design .. of the Rnode hardware with the ESP32 and E22-400M33 RF module.

Is it possible to use this hardware with the 'firmware' from  Markqvist:  https://markqvist.github.io/Reticulum/manual/interfaces.html
In the Netherlands and Belgium we have started the new design pages on http://www.hamesh.eu

Many regards,

PA2EON - Eric

DK9BT, 2025-02-03:
This is free hardware so you can do whatever you want with that.

But i would asume the project you mentioned is not for Ham-Radio cause it includes encryption and thats basicly prohibited on ham frequencies.

But have a look at MeshCom....thats the "normal" firmware which runs on our board and provides mesh comunication for hams

https://icssw.org/meshcom/

https://github.com/icssw-org/MeshCom-Firmware/

karamo, 2025-02-03:
> Is it possible to use this hardware with the 'firmware' from Markqvist: https://markqvist.github.io/Reticulum/manual/interfaces.html In the Netherlands and Belgium we have started the new design pages on http://www.hamesh.eu

durchaus möglich - quite possible

karamo, 2025-02-03:
> the Rnode

Yes, I know Rnode and ported the FW years ago to ESP32.
But I do not give any support to use my PCB for it.

Ja, ich kenne RNode und habe die FW vor Jahren nach ESP32 portiert.
Aber ich gebe keine Unterstützung, um mein PCB dafür einzusetzen.